greatfet-hardware/jasmine #5: load switch indicator dimly illuminated

mossmann, 2019-08-07:
D4 is dimly lit when Jasmine is charging with load switch SW1 turned off.  This happens when charging from Azalea's USB port but not when charging from Jasmine's USB port.  I assume this is due to reverse leakage through D2, but that would be good to test.

mossmann, 2019-08-17:
If this issue persists with D2 removed, it might be caused by leakage from VBAT.

yhetti, 2019-10-16:
Hopefully this will be fixed by adding the LDO. (issue #15)

mossmann, 2020-01-30:
We measured current through LED at 0.5 mA (rev3). Raising the forward voltage of the LED by changing it from red to yellow or green may help.

mossmann, 2020-02-11:
Removing D2 did stop D4 from turning on dimly, so this seems to be caused by reverse leakage through D2.